JonnyBooker/split-flap #1: LinkedList

Lo0kadistraction, 2023-10-29:
ESPMaster will not compile due to error with LinkedList I have installed the linked list from the library manger. https://github.com/ivanseidel/LinkedList That did not seem to help. Please advise. Error message to follow

Arduino: 1.8.18 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from Z:\Projects\3D_Printer\Split_Flap\split-flap-master\ESPMaster\ESPMaster.ino:15:

C:\Users\guita\OneDrive\Documents\Arduino\libraries\LinkedList/LinkedList.h:26:7: error: redeclared with 1 template parameter

   26 | class LinkedList{

      |       ^~~~~~~~~~

In file included from C:\Users\guita\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:29,

                 from Z:\Projects\3D_Printer\Split_Flap\split-flap-master\ESPMaster\ESPMaster.ino:10:

C:\Users\guita\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:39:7: note: previous declaration 'template<class T, template<class> class Item> class LinkedList' used 2 template parameters

   39 | class LinkedList {

      |       ^~~~~~~~~~

In file included from Z:\Projects\3D_Printer\Split_Flap\split-flap-master\ESPMaster\ESPMaster.ino:15:

C:\Users\guita\OneDrive\Documents\Arduino\libraries\LinkedList/LinkedList.h:113:27: error: invalid use of incomplete type 'class LinkedList<T>'

  113 | LinkedList<T>::LinkedList()

      |                           ^

In file included from C:\Users\guita\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:29,

                 from Z:\Projects\3D_Printer\Split_Flap\split-flap-master\ESPMaster\ESPMaster.ino:10:

C:\Users\guita\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:39:7: note: declaration of 'class LinkedList<T>'

   39 | class LinkedList {

      |       ^~~~~~~~~~

In file included from Z:\Projects\3D_Printer\Split_Flap\split-flap-master\ESPMaster\ESPMaster.ino:15:

C:\Users\guita\OneDrive\Documents\Arduino\libraries\LinkedList/LinkedList.h:126:28: error: invalid use of incomplete type 'class LinkedList<T>'

  126 | LinkedList<T>::~LinkedList()

      |                            ^

In file included from C:\Users\guita\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:29,

                 from Z:\Projects\3D_Printer\Split_Flap\split-flap-master\ESPMaster\ESPMaster.ino:10:

C:\Users\guita\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:39:7: note: declaration of 'class LinkedList<T>'

   39 | class LinkedList {

      |       ^~~~~~~~~~

In file included from Z:\Projects\3D_Printer\Split_Flap\split-flap-master\ESPMaster\ESPMaster.ino:15:

C:\Users\guita\OneDrive\Documents\Arduino\libraries\LinkedList/LinkedList.h:145:46: error: invalid use of incomplete type 'class LinkedList<T>'

  145 | ListNode<T>* LinkedList<T>::getNode(int index){

      |                                              ^

In file included from C:\Users\guita\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:29,

                 from Z:\Projects\3D_Printer\Split_Flap\split-flap-master\ESPMaster\ESPMaster.ino:10:

[... 510 lines cut ...]
ServiceFlapFunctions:14:58: error: 'class LinkedList<String>' has no member named 'size'

   14 |       for (int linesIndex = 0; linesIndex < messageLines.size(); linesIndex++) {

      |                                                          ^~~~

ServiceFlapFunctions:15:35: error: no match for 'operator[]' (operand types are 'LinkedList<String>' and 'int')

   15 |         String line = messageLines[linesIndex];

      |                                   ^

ServiceFlapFunctions:23:40: error: 'class LinkedList<String>' has no member named 'size'

   23 |         if (linesIndex <= messageLines.size()) {

      |                                        ^~~~

exit status 1

no matching function for call to 'LinkedList<ScheduledMessage>::LinkedList()'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

chris-guthrie, 2023-10-29:
The problem is related to LinkedList library clashing with ESPAsyncWebServer, you can see a work around listed on this issue report.

https://github.com/ivanseidel/LinkedList/issues/40

Lo0kadistraction, 2023-10-30:
That's a good fix include the LList.h and refactor all LinkedList references to LList and its good to go. thank you!

closed 2023-10-30 by Lo0kadistraction

JonnyBooker, 2024-02-07:
@Lo0kadistraction I am so so sorry! I never saw this get opened as an issue! I will try pay more close attention. Hope the rest of the build went more smoothly for you

@stokes776 I'm glad you solved your issue. The way I got around it was by modifying the original ESPAsyncWebServer library and including it as a zip here in this repository. The wording I used for including the libraries I can look to improve in the readme.

JonnyBooker, 2024-03-03:
@chris-guthrie - Thank you for your link for a workaround. I'm going to do a release in the near future which will include this change so I will include you on the release notes as a thank you :)