JonnyBooker/split-flap #5: Split flap units

Olli1887, 2024-02-29:
Hi is there a minimum of 10 units or can I also use 8 units?

beroliv, 2024-03-02:
Yes you can build it with just 8 Units.  Just follow the instruction and Place the Number of Units in the ESPmaster File.
And you have to build a case for 8 Units. Would be nice you place the Case then on Printable as Remix so others can use it also!

JonnyBooker, 2024-03-03:
Hello! As @beroliv has mentioned, you can do less then 10 units and you can do a maximum amount of units with the current design up to 16 units (this is the limitation of the dip switch, you can go above it but it would need circuit board modification which I think someone has done on printables, link below). 

Going up or down the amount of units means you will have to modify the casing, some people have already done that on printables as remixes:
https://www.printables.com/model/388581-split-flap-display-5-characters
https://www.printables.com/model/465863-split-flap-display-8-characters
https://www.printables.com/model/689265-split-flap-display-12-characters
https://www.printables.com/model/738534-split-flap-display-21-characters

Going to close this one for now but feel free to re-open or continue the conversation if want to know more info

closed 2024-03-03 by JonnyBooker