JonnyBooker/split-flap #6: Wifi Modul Connection issue

beroliv, 2024-03-02:
I had all the time Problems connect the Modul with my Wifi. 

I fix this issue with a External Antenna. I just solder a 62mm wire on the Modul.

See also here: https://www.stall.biz/project/verbesserte-wlan-konnektivitaet-mit-externen-antennen-fuer-wiffi-weatherman-und-andere-module-mit-esp8266/

JonnyBooker, 2024-03-03:
Hi there! Sorry to hear about issues you've been having. This is the first time I'm hearing about any WiFi issues, thank you for providing a link to help with a solution! I will include your link as a note in the readme. I'll close this item once I've done that

JonnyBooker, 2024-03-03:
@beroliv - I've added your link to the common problems section of the readme. I will include you in on the release notes for the next release too. Thank you :)

closed 2024-03-03 by JonnyBooker

beroliv, 2024-03-04:
as i see in google it seems a very common Problem with the ES8266 to connect to asus routers. Interesting is, Prusa Printer use the exact the same module for there Printers. They work  with out any issue.

beroliv, 2024-10-26:
The issue also depended on which access point the ESP connected to. Upstairs, it worked well with the antenna, but in the living room, where the display is located, it didn’t work at all. The solution was:I've identified the issue that was preventing the ESP32 from connecting to my ASUS ZenWiFi XT9. Neither a static IP nor an external antenna solved the problem. I had to disable "Auto Channel" in the settings and set it to a fixed channel (6) instead.