JonnyBooker/split-flap #7: Possible to add static ip?

beroliv, 2024-03-04:
Possible to add static ip to the sketch?

JonnyBooker, 2024-03-05:
I would probably say this is a job for your internet router? Think this would be best to avoid conflicts on your network if multiple devices are trying to reserve the same IP address, as your router is aware of all the devices on the network.

Is there a specific reason you'd like to set it up on the device? Maybe ease or your router doesn't allow such actions?

I've never tried to set up a static IP on an individual Sketch before so cannot confirm if it is possible. The library being used underneath is [WiFiManager](https://github.com/tzapu/WiFiManager) to look after WiFi connections, this might expose something to be able to achieve a static IP. I can have a look into it potentially if get some time over the next few days.

beroliv, 2024-03-07:
Hart Code Static ip Adress with try and error
`//Initialize WiFi
void initWiFi() {
  int wifiConnectTimeoutSeconds = 180;
  WiFi.setPhyMode(WIFI_PHY_MODE_11G); // Connection Problem solved?
  WiFi.mode(WIFI_STA);
  
// Set your Static IP address
IPAddress local_IP(192, 168, 0, 81);
// Set your Gateway IP address
IPAddress gateway(192, 168, 0, 1);

IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8);   //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional
// Configures static IP address
  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
    Serial.println("STA Failed to configure");
  }
  
  
#if WIFI_SETUP_MODE == AP
  SerialPrintln("Setting up WiFi AP Setup Mode");

  wifiManager.setTitle("Split-Flap Setup");
  wifiManager.setHostname("Split-Flap");
  wifiManager.setDarkMode(true);
  wifiManager.setShowInfoUpdate(false);
  wifiManager.setConfigPortalBlocking(true);
  wifiManager.setConfigPortalTimeout(wifiConnectTimeoutSeconds);
  wifiManager.setConnectTimeout(120);    
  wifiManager.setWiFiAutoReconnect(true);
  wifiManager.setSaveConfigCallback([]() {
    //Sadly, if we've had to open up the WiFi manager portal to set the WiFi configuration up
    //then a soft reset is necessary. There looks to be issues with the ESP and running
    //a webserver alongside the WiFiManager. Some reading around this issue here suggests it is
    //not possible to fix:
    //https://github.com/tzapu/WiFiManager/issues/1579
    
    //Suggestion to fix is reset the device:
    //https://github.com/rancilio-pid/clevercoffee/issues/323#issuecomment-1587344185

    SerialPrintln("New WiFi configuration saved. Will need to reboot device to let webserver work...");
    isPendingReboot = true;
  });
  
  //Set the menu options
  std::vector<const char *> menu = { "wifi", "info" ,"param", "sep", "restart", "exit" };
  wifiManager.setMenu(menu);

  SerialPrintln("Attempting to connect to WiFi... Will fallback to AP mode to allow configuring of WiFi if fails...");
  if(wifiManager.autoConnect("Split-Flap-AP")) {
    SerialPrint("Successfully Connected to WiFi. IP Address: ");
    SerialPrintln(WiFi.localIP());

    isWifiConfigured = true;
  }
  
#else
  SerialPrintln("Setting up WiFi Direct");

  if (ssid != "" && password != "") {
    int maxAttemptsCount = 0;
    
    WiFi.begin(wifiDirectSsid, wifiDirectPassword);
    SerialPrint("Connecting");

    while (WiFi.status() != WL_CONNECTED && maxAttemptsCount != wifiConnectTimeoutSeconds) {
      if (maxAttemptsCount % 10 == 0) {
        SerialPrint('\n');
      }
      else {
        SerialPrint('.');
      }

      delay(1000);

      maxAttemptsCount++;      
    }

    //If we reached the max timeout
    if (maxAttemptsCount != wifiConnectTimeoutSeconds) {
      SerialPrint("Successfully Connected to WiFi. IP Address: ");
      SerialPrintln(WiFi.localIP());

      isWifiConfigured = true;
    }
  }

#endif
}`

JonnyBooker, 2024-03-10:
@beroliv - Have you tried the above code and found that it works? I haven't had the chance/capacity to look into setting a static IP address using WiFi Manager recently, will hopefully soon.

beroliv, 2024-03-10:
I used one of the left over esp. it works somehow. But in the List of devices, in the router shows me the wrong old dhcp adress. When i try to open the ui i can open the ui with the programmed adress. Thats strange i know.

JonnyBooker, 2024-03-31:
I've had some time to work on this project. I've introduced a new section for setting a static IP address as a experimental feature.

Currently working off of this feature branch until done further testing: [feature/indefinite-scheduled-messages-and-wifi](https://github.com/JonnyBooker/split-flap/tree/feature/indefinite-scheduled-messages-and-wifi)

Inside the `ESPMaster.ino`, at the top of the file, you'll find a new property you can set to `true`:
```c++
#define WIFI_STATIC_IP      false
```

Then further down the file you'll find another section for variables which you can set your static IP address amongst other settings:
```c++
#if WIFI_STATIC_IP == true
//Static IP address for your device. Try take care to not conflict with something else on your network otherwise
//it is likely to not work
IPAddress wifiDeviceStaticIp(192, 168, 1, 100);

//Your router details
IPAddress wifiRouterGateway(192, 168, 1, 1);
IPAddress wifiSubnet(255, 255, 0, 0);

//DNS Entry. Default: Google DNS
IPAddress wifiPrimaryDns(8, 8, 8, 8);
#endif
```

My initial testing of this has worked, as noted, if you use a IP address of an existing device on your network, it probably isn't going to work. But, looks to work in both direct/ap mode too.

Hopefully this might be what was looking for 😃

JonnyBooker, 2024-04-10:
Will close this issue within the next week, I've created a solution I've tested above which has worked from my testing so feel the original issue has been resolved. I will merge the associated PR within the next week and update the `readme.md` to have details on how to use the new feature.

JonnyBooker, 2024-04-18:
Closing this item down now the PR has been merged. Static IP Address functionality is in `master` and in release [2.2.0](https://github.com/JonnyBooker/split-flap/releases/tag/v2.2.0)

closed 2024-04-18 by JonnyBooker