JonnyBooker/split-flap #10: Compatability with Home Assistant?

skynet01, 2024-04-18:
Would love to dive into this project but wanted to check to see if it's possible to make it compatible with [Home Assistant](https://github.com/home-assistant/core)? Maybe Split-Flap can receive messages on MQTT protocol? Or have a version of this firmware running on [ESPHome](https://esphome.io/index.html)? 

The use case is that with Home Assistant it makes it REALLY easy to automate this for messages, notifications, sensor information etc.

skynet01, 2024-04-18:
Found an mqtt  version here: https://github.com/nctiggy/split-flap-mqtt maybe they can be merged?

JonnyBooker, 2024-04-18:
@skynet01 - Hello! You beat me to sending that link! 

It is something I've considered doing but personally didn't have much use for myself as I don't run a HomeAssistant instance at home so left the idea on the backburner in favour of other projects.

It is something I can potentially look into if I manage to get some time in the coming weeks, from what I've seen the implementation doesn't look too bad, and happy to post back here if I am making progress on it.

Happy to consider a PR potentially otherwise if capable!

N1c093, 2024-07-10:
I'm considering building this display as well. I use Home Assistant as well.

Is there currently no API available to transfer text to the display? Is it only possible through the web interface?

skynet01, 2024-07-10:
You would have to use MQTT integration to send messages to the Split Flap

JonnyBooker, 2024-07-11:
@N1c093 - There is an "API" of sorts, you can submit a POST body to this [webserver endpoint](https://github.com/JonnyBooker/split-flap/blob/671666c3b1b0ad3e78ba12a19e038cdceb33cafa/ESPMaster/ESPMaster.ino#L328) and it will update the display, however it is geared around serving the web interface so the action at the end is to redirect

But @skynet01 is correct, the typical implementation is via MQTT and that is the implementation that is followed [here](https://github.com/nctiggy/split-flap-mqtt). I haven't got the capacity at the moment to be able to look into it unfortunately at this time, apologies. Not saying never just not at the moment.

JonnyBooker, 2024-10-16:
I'm closing this issue for now as we have established it is not currently implemented. In future I may introduce a "Proper" API which can be used alongside the current implementation.

This feature will become evident through the releases :)

closed 2024-10-16 by JonnyBooker