JonnyBooker/split-flap #11: That W...

ThatWileyGuy, 2024-05-08:
[image]

[image]

... does anyone know what font was used to make these cards?

ThatWileyGuy, 2024-05-09:
I found another implementation and quickly hacked this up:
```
letter_offset_x = -0.5;
letter_offset_y = -24;
letter_font = "Consolas:style=Bold";
letter_height = 0.7;
letter_width = 1.0;
letter_thickness = 0.4;
flap_height = 35;
flap_width = 43;
flap_thickness = 1.0;
letter_color = [1, 1, 1]; // white
flap_color = [0, 0, 0]; // black
flap_gap = 0;
flap_pin_width = 2;
flap_pin_offset = 1.2;
flap_pin_thickness = 1.2;
flap_drum_cut_length = 15.6;
flap_offset = 0.35;

module flap_2d() {
    translate([0, flap_offset, 0])
        difference() {
            square([flap_width, flap_height]);
            union() {
                square([flap_pin_width, flap_pin_thickness]);
                translate([0, flap_pin_offset + flap_pin_thickness, 0])
                    square([flap_pin_width, flap_drum_cut_length]);
                translate([flap_width - flap_pin_width, 0, 0])
                    square([flap_pin_width, flap_pin_thickness]);
                translate([flap_width - flap_pin_width, flap_pin_offset + flap_pin_thickness, 0])
                    square([flap_pin_width, flap_drum_cut_length]);
            }
        }
}

module flap() {
    color(flap_color)
    translate([0, 0, -flap_thickness/2])
    linear_extrude(height=flap_thickness) {
        flap_2d();
    }
}

module draw_letter(letter) {
    scale([letter_width, 1, 1])
        translate([letter_offset_x, letter_offset_y, 0])
            text(text=letter, size=flap_height * letter_height * 2 + flap_offset * 2, font=letter_font, halign = "center");
}

module flap_letter(letter, thickness, half = 0) {
    color(letter_color)
     {
        if (half != 0) {  // trimming to top (1) or bottom (2)
            rotate([180, 0, 0])
            translate([0, 0, flap_thickness/2 - letter_thickness])
            linear_extrude(height=thickness)
                intersection() {
                    union() {
                        if (half == 2) {
                            flap_2d();
                        } else if (half == 1) {
                            rotate([180, 0, 0])
                                flap_2d();
                        }
                    }
                    translate([flap_width/2, 0, 0])
                        draw_letter(letter);
                }
        } else {
            translate([0, 0, flap_thickness/2 - letter_thickness])
            linear_extrude(height=thickness)
                intersection() {
                    union() {
                        flap_2d();
                        rotate([180, 0, 0])
                            flap_2d();
                    }
                    translate([flap_width/2, 0, 0])
                        draw_letter(letter);
                }
        }
    }
}

module make_card_pair(letter1, letter2, letter3)
{
    difference() {
        union() {
            flap();
            rotate([180, 0, 0])
                flap();
        }
        
        flap_letter(letter1, letter_thickness + 0.1, 2);
        flap_letter(letter2, letter_thickness + 0.1);
        flap_letter(letter3, letter_thickness + 0.1, 1);
        
    }
    flap_letter(letter1, letter_thickness, 2);
    flap_letter(letter2, letter_thickness);
    flap_letter(letter3, letter_thickness, 1);

}

letters = " ABCDEFGHIJKLMNOPQRSTUVWXYZ$&#0123456789:.-?!";

for(i = [0:2:len(letters)-1]) {
    last = i == 0 ? len(letters) - 1 : i - 1;
    next = i == len(letters) - 1 ? 0 : i + 1;
    
    translate([i/2 * (flap_width + 5), 0, 0])
    make_card_pair(letters[last], letters[i], letters[next]);
}

```

... which does render all the cards successfully (although it makes an extra card that's a duplicate of the first one).
Still trying to figure out how to get from here to multimaterial STLs. I have no idea what I'm doing on OpenSCAD (I normally use Fusion 360), so feel free to lecture me about how this should be done.

JonnyBooker, 2024-05-10:
@ThatWileyGuy - Hello! I know... It bothers me too but after assembling so many flaps wheels, I can't be bothered to correct it :D 

Nice work on rendering those cards, pretty cool. Just as a note, someone posted a remix of the increased size "W" here:
https://www.printables.com/model/481904-larger-height-w-flaps

Just thinking out loud as an alternative, I think in Bambu slicer you can just maybe grab the "W" letter in the assembly and stretch it maybe.

ThatWileyGuy, 2024-05-10:
I ended up finishing the openscad script and writing a powershell script to generate the black and white STLs for each pair of cards. The big benefit is that you can pick your own font and generate cards for it and/or change which characters are in the set of 45.
Do you think it'd be worth adding the scripts to the repo?

JonnyBooker, 2024-05-13:
@ThatWileyGuy - Apologies, I've been away the past few days.

Nice one, yeah sure I'd be happy to have it in the repository when you've got it into a happy state. Want to make sure you have credit for it too so good to do it as a pull request, if you want to create one with the script(s) and maybe add a note on usage in the readme, happy to accept that if sounds good to you?

JonnyBooker, 2024-07-11:
I'm going to close this item as not heard back in a while. If want to contribute the script that you made happy to accept it into the repo and ensure your accredited for that :) 

Re-open this one if want to discuss further.

closed 2024-07-11 by JonnyBooker