JonnyBooker/split-flap #12: Days count up

shagadeli, 2024-05-22:
Any way to make it count the elapsed days up automatically?
The idea is to make a "x days since last accident" kind of thing.

Thanks

JonnyBooker, 2024-05-26:
Hello!
I think it could be possible, the code at the moment as it stands for Countdown will ensure it doesn't go [negative](https://github.com/JonnyBooker/split-flap/blob/671666c3b1b0ad3e78ba12a19e038cdceb33cafa/ESPMaster/ServiceCountdownFunctions.ino#L16), this could probably be changed to make it so that rather then saying "X days" it could be modified to say "X days to go" or "X days past" or something like that and allow it to go "negative".

Something could look at as trial experiment sometime in future maybe (in the next week or two potentially) unless you would like to have a go! Happy to offer advice.

JonnyBooker, 2024-07-16:
I'm going to mark this item as closed as it is not an issue, more a feature that is not implemented. If this feature gets implemented in future I'll be sure to report back here :). Feel free to re-open if feel necessary.

closed 2024-07-16 by JonnyBooker