JonnyBooker/split-flap #13: DCF radio module or a battery-buffered timer

JoeBerlin69, 2024-05-31:
Hello,
I think the SplitFlap display is great. I've already built several for friends and relatives. Most of them just use the clock. Now I have someone who doesn't have WiFi at home.
Couldn't the electronics be expanded to include a DCF radio module or a battery-buffered timer?
Best wishes
Jörg

JonnyBooker, 2024-05-31:
Hello!
That is a lot of patience to do it multiple times! 

I'm afraid I don't know much when it comes to the electronics side of it, it might be worth posting that question to the original creator David whom I forked this project from. I can do the software side of it but less so the hardware/electronic side.

Many Thanks

JoeBerlin69, 2024-06-02:
Hi, i am trying it for myself with a DS3231 Modul. I use the help with ChatGPT and it could work. If possible wr stay in contact during this process and we can help each other. Thanks a lot for your answer.Kind regardsJoergVon meinem/meiner Galaxy gesendet

JonnyBooker, 2024-06-03:
@JoeBerlin69 - Sure, feel free to post here and I'll try answer your questions, no guarantee I can but can try

JonnyBooker, 2024-07-16:
Not heard anything about this in a while @JoeBerlin69, how is it all going?

JonnyBooker, 2024-07-18:
Fair enough, no need for sorry! Happy to pick this back up and discuss if you get anymore time to work on it in future! I'll close this issue for now but happy to re-open it if need be

closed 2024-07-18 by JonnyBooker