JonnyBooker/split-flap #14: Comptability with split-flap-fae-pcb for reduced cost

TwinZA, 2024-06-27:
Hello, 

I apologize if this is the wrong place to ask this question but Im wondering if this project is primarily compatible with https://github.com/codingcatgirl/split-flap-fae-pcb for the reduced cost of not needing the arduino nanos

JonnyBooker, 2024-07-11:
@TwinZA - Apologies, I use my GitHub for work too and think the notifications get lost in a sea of other notifications...

My understanding is that it is a drop in replacement for the original design. It is still using the ESP8266 so the main ESPMaster _should_ work just fine with it. I don't know much about the Atmega328P that it uses but reading [this](https://store-usa.arduino.cc/products/arduino-nano?selectedStore=us) suggests that is the Arduino Nano is based upon it using the same processor, so it should be possible to use this PCB

But I'll caveat that with, I'm not 100% sure as I'm not a board designer by trade ;)

JonnyBooker, 2024-09-09:
Closing this one as haven't heard back for a while. Happy to re-open to discuss further if want more information :)

closed 2024-09-09 by JonnyBooker