JonnyBooker/split-flap #16: Extension to 32 or 64 modules

goguelnikov, 2024-08-23:
Hello
Do you think it would be possible to extend the code and adapt the PCB to be able to link 32 or 64 modules?
Maybe by replacing the dip switch with an auto-detection of the ID?
Best

JonnyBooker, 2024-09-09:
Hello!
Sorry for the delay in replying, been on vacation!

I know of this project someone started as a fork which might be able to get around some of the limitations (https://github.com/codingcatgirl/split-flap-fae-pcb). However I am no PCB designer so I cannot confirm how well it works so you'll have to do your own research on it I'm afraid. I am aware though that David, who did the original work, recommended it originally on his repo, however someone did report having issues with that repository and removed it (https://github.com/davidkingsman/split-flap/commit/14a22d9f38900aa520feeb70656357b8fd719e45)

On printables someone did do a large 21 character unit and would suggest that you can ignore the DIP switches entirely and use the original design of PCB (https://www.printables.com/model/738534-split-flap-display-21-characters) 

Otherwise I don't know much about PCB design to be able help any further! Hopefully the above links will be of use :)

JonnyBooker, 2024-10-16:
I am closing this issue for now as not heard back in a while. Happy to continue this conversation and have it re-opened should you want to discuss this further :)