JonnyBooker/split-flap #18: Reading RSS?

Tomtim1, 2024-09-21:
Hi! Is there any possibility to write using RSS on this screen? I imagine it would work the best with GPT to shorten news everyday to 2-3 lines of text and display them one by one every few seconds.

JonnyBooker, 2024-09-23:
Hello!
I feel that is quite a niche feature for this type of device to do, so I don't think its quite a fit for this project at this time. However I think it should be very possible and happy to outline how think that could be done. 

You could create a new "Mode" similar to date/text/time/countdown which could support it and add in a bunch of configuration options to point it at an RSS Feed. There are several tutorials on reading a RSS Feed on google with a Arduino which I think would be simple to translate and even [this](https://www.arduino.cc/reference/en/libraries/rssread/) library which might help.

You could probably leverage a lot of what is already there as I've already implemented message splitting which if you were to "parse" your text from a RSS feed/ChatGPT, you can pass it along to the `showText` method [here](https://github.com/JonnyBooker/split-flap/blob/671666c3b1b0ad3e78ba12a19e038cdceb33cafa/ESPMaster/ServiceFlapFunctions.ino#L6) which would take care of splitting all the words across several turns of the display. 

I have some concerns about ensuring that you don't pull down from the RSS feed too often and it does not exhaust your ChatGPT tokens so you'd have to be careful on that one to not rack up a bill!

Happy to give feedback and answer questions if decide to pursue this further :)

JonnyBooker, 2024-12-15:
Going to close this item down as won't be working on this item for the near future. I will update here is work on such functionality in future :)

closed 2024-12-15 by JonnyBooker