JonnyBooker/split-flap #19: EEPROM Offset Not Working

jdylanknight, 2024-09-27:
Hi y'all!

I am quite new to this and am working through the offset settings for my first unit.

I am not getting consistent results with the test_enable un-commented in Arduino IDE -- it seems to be hitting random letters and not homing at all. It will spin a few full rotations and stopping on a letter, then executing the following sequence:

  int calLetters[10] = {0, 26, 1, 21, 14, 43, 30, 31, 32, 39};

As I have tested the hall sensor for correct polarity and it functions outside of this test, what else may I keep an eye out for?

Additionally, when setting EEPROM, it is recommended to start with the value of 100. I looked all over and am still unsure what unit this is in or how to adjust it should I see a consistent yet incorrect homed letter I can correct for (e.g., "B" instead of the " " as it should be).

Thank you in advance for your patience and any information y'all may provide. I am otherwise rocking and rolling with this project.

JonnyBooker, 2024-10-16:
We discussed this privately and the issue seems to be related to the Hall Sensors. However it has highlighted it is not obvious what the offset code is doing so will look to update the readme with more information regarding this

jdylanknight, 2024-10-21:
Thanks Jonny! I appreciate your help on my question and will update you once I have tried other hall sensor modules. :-)

JonnyBooker, 2024-12-19:
Closing this item as not heard anything for a while. Please do reach out if need any further help :)

closed 2024-12-19 by JonnyBooker

jdylanknight, 2025-07-20:
Hey @LucaBastelt! Great to meet you.

My issue ended up being resolved after changing the Hall Sensors I was using. I dropped in the new sensor, and they ended up working as planned. Unfortunatly it looks like the sensors you used were the exact sensors that solved my issues I saw:
[This is the set](https://www.aliexpress.us/item/2251832475321023.html?spm=a2g0o.order_list.order_list_main.5.46621802v4zKnX&gatewayAdapt=glo2usa)

LucaBastelt, 2025-07-20:
@jdylanknight Damn, but thank you for checking! Ill have to try around with my magnet placement next, i might have screwed that up by just a mm 😅