JonnyBooker/split-flap #22: Time format

JNSwalmen, 2024-11-18:
Is it possible to add the seconds to the time?
Also I set the display to timezone Berlin but the time is one hour behind. Did i do something wrong ?

JonnyBooker, 2024-12-15:
Sorry for the delay, I've been very busy on other projects.

For seconds, it is possible, you can change the format of the clock that is displayed here:
https://github.com/JonnyBooker/split-flap/blob/671666c3b1b0ad3e78ba12a19e038cdceb33cafa/ESPMaster/ESPMaster.ino#L113

However, I will say that it will not work as well as you expect. It works great while counting up from 0-9, however when it comes back to getting back to 0, the split flaps have to pretty much cycle all of the way through again to get to 0 and this will take a second or two, meaning the display will always be catching up to itself. So use it at your own peril!

I think the timezone needs to be set to "Europe/Berlin" to make it work:
https://github.com/JonnyBooker/split-flap/blob/671666c3b1b0ad3e78ba12a19e038cdceb33cafa/ESPMaster/ESPMaster.ino#L108

It should all work off of IANA codes so should be fine.

JonnyBooker, 2025-01-16:
Closing this item for now as answered regarding seconds and not heard back anything further about the timezone. Please feel free to re-open if have any further questions or want further help :)

closed 2025-01-16 by JonnyBooker