JonnyBooker/split-flap #28: Esp8266 not responding to Dashboard Buttons

RobloxHeckerExE, 2025-06-21:
-Dashboard shows
-No error in HTML
-Units Amount is set correctly
-Hall sensor is connected
-Motor is connected
-At the beginning the split flap calibrates sucessfully

strix214, 2025-07-25:
I met same problem. Without changing the code, sometime I2C just not work, sometime can work(I dont change anything). And I dont know what happen.

strix214, 2025-08-14:
> My suggestions at this time would be to ensure debug flags like `SERIAL_ENABLE` and `UNIT_CALLS_DISABLE` are false in your ESPMaster code. Same would go for the Unit.ino too to ensure those flags are not true, although you say the units calibrate so that one is probably not helpful.
> 
> Other thought is are you dip switches set correctly for each unit so they are addressable via the ESP? There should be some information in the README.
> 
> Any other information can provide would be useful

I have 2 Units now, when I set #define UNITS_AMOUNT  1, 2 Units can work well to display the same letter; If I set #define UNITS_AMOUNT  2, All the Unit can't move. And I am finding what leads to this situation.

pvbrakel, 2025-08-16:
@strix214 I've had this issue too. In my case in unit.ino method getAddress() returned the bits in the wrong order. Thus the second unit was listening to address 8 instead of 1.
I had to change the code to 
`//returns the adress of the unit as int from 0-15
int getaddress() {
  int address = !digitalRead(ADRESSSW1) + (!digitalRead(ADRESSSW2) * 2) + (!digitalRead(ADRESSSW3) * 4) + (!digitalRead(ADRESSSW4) * 8);
  return address;
}`

StreamMovieYT, 2025-08-16:
> > [@strix214](https://github.com/strix214) I've had this issue too. In my case in unit.ino method getAddress() returned the bits in the wrong order. Thus the second unit was listening to address 8 instead of 1. I had to change the code to
> > ```
> > //returns the adress of the unit as int from 0-15
> > int getaddress() {
> >   int address = !digitalRead(ADRESSSW1) + (!digitalRead(ADRESSSW2) * 2) + (!digitalRead(ADRESSSW3) * 4) + (!digitalRead(ADRESSSW4) * 8);
> >   return address;
> > }
> > ```
> 
> Sorry guys my mistake. For some reason I read it as LSB first, while of course it should be MSB first. Ignore me, I should learn to better read the manual 😅

Can u please contact me over discord and help me with my problem, that would be so nice of u

Discord: streammovieyt

closed 2025-09-21 by github-actions[bot]

YingLuoNou, 2025-10-03:
You can try to view issue #31 , where there may be a solution or related information to the problem you are encountering, which may be helpful to you.